atomicobject/heatshrink #33: Java library

cagnulein, 2015-06-23:
Hi, thanks for your great work!
I would like to decompress the data generated by an embedded system on Java.
Have you got a ready-to-use solution?
Should i use a standard lzss java class?

Thanks

silentbicycle, 2015-06-23:
I don't have an implementation I've checked for compatibility, sorry. (I don't do much work on the JVM.)

LZSS is a pretty standard algorithm, and most of the unique implementation details for heatshrink either have to do with working incrementally or specific heuristics for which patterns are / aren't an improvement on literal output. The decoder does far less work than the encoder, and a blocking Java decoder implementation could be significantly simpler.

I may write an implementation eventually (which would be tested against the current library version), but I'd be happy to include an implementation with a compatible license.

cagnulein, 2015-06-23:
Ok thanks. In the meantine i'm compiling a dll version of your decoder in order to linking in Java.
I will let you know if it will ok.

cagnulein, 2015-06-23:
I confirm you that linking the c code compiled as a dll and linking it from java works like a charm.

silentbicycle, 2015-06-23:
Fantastic. :)

I added issue #34, to have a compatible simple, blocking implementation as documentation, to make porting like this easier.

closed 2015-06-23 by silentbicycle

markrileybot, 2016-12-03:
In case anyone is still interested, I started working on this today: https://github.com/markrileybot/heatshrink-java/

Decoding works fine.  I'll write an encoder soon.

markrileybot, 2016-12-11:
Java encoder/decoder complete

https://github.com/markrileybot/heatshrink-java/