atomicobject/heatshrink #34: Add blocking example implementation of LZSS

silentbicycle, 2015-06-23:
This would document some implementation details, and help people who want to make output-compatible implementations in other languages such as Java ( See issue #33 ).

LZSS is quite a bit simpler if the implementation can run in one pass -- much of heatshrink's code is managing the suspending/resuming.

markrileybot, 2016-12-11:
Java port:

https://github.com/markrileybot/heatshrink-java/

ecm-pushbx, 2018-12-15:
I considered implementing a decoder for heatshrink but the implementation is too complicated for me to figure out. My decoder would work with a buffer that's usually large enough to hold both the compressed and decompressed data. (I also implement overlapping buffers when possible, but that's just an extra.)

LZ4 has some excellent documentation of their format, which I used to implement my decoder. For reference, their documentation is at https://github.com/lz4/lz4/blob/e5a1911ec298a03242b2c8f954f3b511d5a743f6/doc/lz4_Block_format.md and https://github.com/lz4/lz4/blob/e5a1911ec298a03242b2c8f954f3b511d5a743f6/doc/lz4_Frame_format.md

And my project for these decoders is at https://bitbucket.org/ecm/inicomp/src/default/