atomicobject/heatshrink #47: Compiling for ESP8266 fails

AleriDom, 2017-10-04:
Hey,

I've just tried to use the library on a ESP8266 platform. Sadly there are several errors display:

Error	GCC	invalid conversion from 'void*' to 'io_handle*' [-fpermissive]	121:37
Error	GCC	in expansion of macro 'HEATSHRINK_ERR'	144:9
Error	GCC	in expansion of macro 'HEATSHRINK_ERR'	176:28
Error	GCC	in expansion of macro 'HEATSHRINK_ERR'	179:38
Error	GCC	in expansion of macro 'HEATSHRINK_ERR'	214:30
Error	GCC	in expansion of macro 'HEATSHRINK_ERR'	229:34
Error	GCC	'heatshrink_encoder_alloc' was not declared in this scope	282:86
Error	GCC	in expansion of macro 'HEATSHRINK_ERR'	303:26
Error	GCC	'heatshrink_encoder_free' was not declared in this scope	305:32

Any help is appreciated :)

AleriDom, 2017-10-04:
first may be fixed as:

from:
    io = malloc(sizeof(*io) + buf_sz);
to:
    io = (io_handle*)malloc(sizeof(*io) + buf_sz);

silentbicycle, 2017-10-04:
`heatshrink.c` is a command line program for working with heatshrink-compressed data. You probably won't need to compile it for your ESP8266 at all -- it probably just needs the encoder and/or decoder files. Does everything else compile without issue?

AleriDom, 2017-10-04:
right. removed heatshrink.c from project, tho still some undefined references but the encoder and decoder header files are included:

.pioenvs\nodemcuv2\src\main.o:(.text._Z17heatshrink_encodePhjPFvhE+0x20): undefined reference to `heatshrink_encoder_poll(heatshrink_encoder*, unsigned char*, unsigned int, un

signed int*)'
.pioenvs\nodemcuv2\src\main.o: In function `heatshrink_encode(unsigned char*, unsigned int, void (*)(unsigned char))':
main.cpp:(.text._Z17heatshrink_encodePhjPFvhE+0x4a): undefined reference to `heatshrink_encoder_reset(heatshrink_encoder*)'
main.cpp:(.text._Z17heatshrink_encodePhjPFvhE+0x6b): undefined reference to `heatshrink_encoder_sink(heatshrink_encoder*, unsigned char*, unsigned int, unsigned int*)'
main.cpp:(.text._Z17heatshrink_encodePhjPFvhE+0x8e): undefined reference to `heatshrink_encoder_finish(heatshrink_encoder*)'
main.cpp:(.text._Z17heatshrink_encodePhjPFvhE+0xa3): undefined reference to `heatshrink_encoder_poll(heatshrink_encoder*, unsigned char*, unsigned int, unsigned int*)'
main.cpp:(.text._Z17heatshrink_encodePhjPFvhE+0x10e): undefined reference to `heatshrink_encoder_finish(heatshrink_encoder*)'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\nodemcuv2\firmware.elf] Error 1

silentbicycle, 2017-10-04:
Are you sure `heatshrink_encoder.o` is being linked? Those undefined references are all the non-static functions it defines.

AleriDom, 2017-10-04:
OK, seems to be a problem with the build environment, thanks.

closed 2017-10-04 by AleriDom