atomicobject/heatshrink #48: Compilation Error:  looking for <err.h>

arnoldvillasanta, 2017-10-08:
/Arduino/libraries/heatshrink-master/heatshrink.c:32:17: 
fatal error: err.h: No such file or directory
#include <err.h>
             
Where can I download err.h?

silentbicycle, 2017-10-08:
What OS are you compiling on?

arnoldvillasanta, 2017-10-09:
MacOS

Doom4535, 2018-05-10:
Was a solution to this ever found?

silentbicycle, 2018-07-31:
You should already have `err.h` as `/usr/include/err.h`, but the build failure may have more to do with building in an Arduino context -- the Arduino platform is based on C++, not C, which impacts the build setup in many ways.

Are you building with the included makefile? `heatshrink.c` is the main source for the command line tool, which really doesn't make sense to build for an Arduino, but has built without issue on macOS. I haven't tried since Yosemite, though.

If that's not enough to get you unstuck, I need more details about how you're trying to build.

commit 3d89d60e (2021-07-04): Add comment about heatshrink.c builds breaking due to err.h or getopt.h.

A couple people have opened issues because they can't get this file
to build for Arduinos, etc., but it isn't supposed to be. Add a comment
comment explaining why right next to those header includes to clarify.

Closes #48, #61.