atomicobject/heatshrink #67: Decoder for JavaScript?

slaff, 2021-07-02:
I would like to encode with heatsink a bitmap image data on a micro-controller and display it in the browser.  
Therefore my question is: is there a JavaScript implementation of the decoder?

BenBE, 2021-07-02:
Currently not, but you should be able to import the decoder source with little changes using [Emscripten](https://emscripten.org/docs/porting/connecting_cpp_and_javascript/Interacting-with-code.html).

silentbicycle, 2021-07-04:
Most of the complexity is to account for suspending/resuming on IO. Otherwise, it's a pretty straightforward implementation of LZSS (https://en.wikipedia.org/wiki/LZSS). I have been meaning to add a one-pass, blocking reference implementation in C that could be used for porting, so I will prioritize that for the 0.5.0 release.

I'm not aware of a Javascript implementation, and I'm not proficient with JS myself, but if somebody writes one with appropriate licensing we could include it in a `contrib/` directory.

slaff, 2021-07-05:
Thank you all for your answers. I have found the following project: https://github.com/iotile/heatshrink-ts.git which is a TypeScript implementation of the decoding. Using `tsc`, an app that converts TypeScript  to JavaScript, I was able to get a JavaScript version of the decoder that worked like charm. The unpolished but working JS file is attached to this post. To use it from JavaScript the following minimum code sample can be used:

```html
<script src="heatshrink.js"></script>
<script>
	var decoder = new HeatshrinkDecoder(10, 4, encodedInput.byteLength);
	decoder.process(encodedInput);
	var data = decoder.getOutput();
</script>
```

In the constructor 10 and 4 should be replaced with the windowBits and lookaheadBits values used in the encoder to encode the content.

[heatshrink.zip](https://github.com/atomicobject/heatshrink/files/6762392/heatshrink.zip)

closed 2021-07-07 by slaff