atomicobject/heatshrink #70: [Question] Outpud compat of static vs dynamic version

BenBE, 2021-09-29:
Is the output of a stream of data encoded with the static implementation compatible with the same data stream using the encoder with dynamic allocation (while using same parameters)?

In particular, when compressing a stream with the static version using parameters for `HEATSHRINK_STATIC_WINDOW_BITS` and `HEATSHRINK_STATIC_LOOKAHEAD_BITS` do I simply need to mirror those exact same settings with the dynamic version and decompression should be fine or is there anything else to take care of? What is the process in the other direction when compressing data with the dynamic version: What do I need to take care of such that a given static configuration can decompress the data?

Are there in general any overall guarantees about stream compatibility based on the settings done in the configuration header?

UnePierre, 2021-09-29:
In our project, we compress dynamically on a PC and decompress with (the same) static settings on a bare-metal micro-controller. In another project, we compress on the micro-controller with (other) static settings and decompress on the PC.

In either direction, never had any problems.

silentbicycle, 2021-09-29:
Static and dynamic builds with the same parameters should be compatible. If they aren't, it's a bug. The only difference is whether the underlying data structures are dynamically allocated or statically allocated with a fixed size.

closed 2021-09-29 by BenBE