atomicobject/heatshrink #84: compress stream of array involve negative values

zaiafat, 2024-02-17:
thank you for good library
I have a large uint8 data (sound data). I want to compress it with heatshirink.
for examle assume I want to compress this 240 byte
const char data[] =	
{136, 133, 128, 117, 114, 144, 136, 130, 122, 107, 138, 139, 130, 128, 104, 128, 139, 132, 133, 104,
 120, 135, 132, 141, 110, 114, 136, 129, 145, 119, 104, 132, 130, 150, 130, 98 , 120, 122, 151, 145,
 99 , 114, 110, 141, 162, 108, 111, 107, 126, 168, 120, 111, 110, 116, 171, 136, 111, 114, 110, 162,
 150, 108, 111, 107, 151, 165, 119, 110, 107, 138, 169, 130, 108, 105, 122, 168, 142, 108, 113, 110,
 153, 154, 108, 111, 111, 144, 169, 123, 110, 105, 126, 174, 139, 111, 105, 108, 160, 153, 111, 99 ,
 99 , 150, 169, 125, 98 , 89 , 125, 174, 142, 101, 92 , 105, 160, 162, 108, 98 , 105, 147, 174, 120,
 93 , 122, 174, 148, 102, 98 , 107, 150, 168, 122, 98 , 107, 126, 165, 150, 104, 107, 116, 151, 174,
 101, 102, 119, 168, 150, 105, 102, 110, 150, 172, 125, 95 , 104, 129, 171, 148, 93 , 92 , 113, 156,
 119, 90 , 105, 129, 169, 148, 98 , 99 , 113, 136, 159, 122, 99 , 116, 114, 139, 145, 104, 111, 117,
 156, 138, 107, 113, 104, 132, 166, 129, 104, 99 , 107, 163, 171, 119, 101, 90 , 126, 181, 144, 102,
 104, 160, 181, 126, 98 , 101, 116, 165, 165, 114, 107, 107, 125, 169, 147, 117, 117, 104, 139, 163,
 113, 105, 113, 156, 168, 128, 108, 110, 120, 166, 156, 104, 104, 102, 132, 174, 139, 108, 101, 102}; 
the compress function dont have good result with this raw data.
the dalta of the data also have negative values also dont have good compression in result.

signed char test_data[256];
for(int i=0;i<240;i++) { test_data[i]=data[i+1]-data[i];} 

is the heatshirink good choice for my aim?
is it possible give me an example with such above values?
thanks in advance

silentbicycle, 2024-02-20:
Heatshrink works by finding repeated patterns/substrings and emitting backward references rather than repeating them (using [LZSS](https://en.wikipedia.org/wiki/Lempel%E2%80%93Ziv%E2%80%93Storer%E2%80%93Szymanski)), so it's better suited to compressing things like text or bitmaps. Numerical data that doesn't repeat exactly but has statistical patterns (sound waves, sensor data, etc.) will probably compress more effectively using methods like [arithmetic coding](https://en.wikipedia.org/wiki/Arithmetic_coding).

BenBE, 2024-02-21:
As you mentioned that your source is some sort of audio you may find [FLAC](https://en.wikipedia.org/wiki/FLAC) may work better for your purpose.