atomicobject/heatshrink #89: Encoding with options -w 10 -l 5 and then decoding leads to corrupted file

BramEPC, 2024-06-03:
I did a quick test to evaluate heatshrink compressing a GPX file with parameters -w 10 -l 5 but the decoded file is smaller than the original and is has several zeroed out  parts.
 
I got this result both by compiling heatshrink under linux as by using pip install heatshrink2 under windows:
./heatshrink -e -w 10 -l 5 test.gpx  test.hs
./heatshrink -d test.hs test_decoded.gpx

and

python -m heatshrink2 compress -w 10 -l 5 test.gpx test.gpx_hspython_w10_l5
python -m heatshrink2 decompress test.gpx_hspython_w10_l5 test.gpx_decompress_hspython_w10_l5

If I do not specify the -w/-l parameters under python then the result is correct. 
Unless I am missing something in the documentation, w 10 / l 5 should be a valid parameter combination (w between 4 and 15 and l between 3 and 10-1), so I assume I am running into a bug ?

Test files in attachment:
[test.zip](https://github.com/user-attachments/files/15540512/test.zip)

BramEPC, 2024-06-03:
I realized the problem : when decoding, you ALSO have the specify the parameters used during the encoding. 
That was not clear to me from the documentation.

closed 2024-06-03 by BramEPC

silentbicycle, 2024-06-04:
OK.